jaseg/gerbolyze #2: opencv error

XenGi, 2019-03-25:
```
$ ls -l
total 4.6M
-rw-rw-r-- 1 xengi xengi  32K 2018-06-28 22:14:12 002.png
-rw-rw-r-- 1 xengi xengi 2.9M 2015-12-11 01:34:57 CX_4800304000.stp
drwxrwxr-x 2 xengi xengi 4.0K 2019-03-24 17:58:12 HIROSE_CX70M-24P1.pretty/
-rw-rw-r-- 1 xengi xengi 1.6K 2019-03-26 00:16:00 README.md
-rw-rw-r-- 1 xengi xengi  27K 2019-03-26 00:16:52 darling-B.Cu.gbr
-rw-rw-r-- 1 xengi xengi  23K 2019-03-26 00:16:52 darling-B.Mask.gbr
-rw-rw-r-- 1 xengi xengi  494 2019-03-26 00:16:52 darling-B.Paste.gbr
-rw-rw-r-- 1 xengi xengi 8.0K 2019-03-26 00:16:52 darling-B.SilkS.gbr
-rw-rw-r-- 1 xengi xengi 1.1K 2019-03-26 00:16:52 darling-Edge.Cuts.gbr
-rw-rw-r-- 1 xengi xengi  89K 2019-03-26 00:16:52 darling-F.Cu.gbr
-rw-rw-r-- 1 xengi xengi  54K 2019-03-26 00:16:52 darling-F.Mask.gbr
-rw-rw-r-- 1 xengi xengi  50K 2019-03-26 00:16:52 darling-F.Paste.gbr
-rw-rw-r-- 1 xengi xengi  14K 2019-03-26 00:16:52 darling-F.SilkS.gbr
-rw-rw-r-- 1 xengi xengi  275 2019-03-26 00:23:26 darling-NPTH.drl
-rw-rw-r-- 1 xengi xengi  862 2019-03-26 00:23:26 darling-PTH.drl
-rw-rw-r-- 1 xengi xengi 5.7K 2019-03-26 00:02:27 darling-cache.lib
-rw-rw-r-- 1 xengi xengi  11K 2019-03-24 17:58:38 darling.bak
-rw-rw-r-- 1 xengi xengi  79K 2019-03-26 00:22:43 darling.kicad_pcb
-rw-rw-r-- 1 xengi xengi  73K 2019-03-26 00:18:07 darling.kicad_pcb-bak
-rw-rw-r-- 1 xengi xengi  13K 2019-03-26 00:22:27 darling.net
-rw-r--r-- 1 xengi xengi 1.1K 2019-03-26 00:22:41 darling.pro
-rw-rw-r-- 1 xengi xengi  11K 2019-03-26 00:02:26 darling.sch
-rw-rw-r-- 1 xengi xengi 289K 2019-03-26 00:03:37 darling_3d.png
-rw-rw-r-- 1 xengi xengi  89K 2019-03-26 00:01:53 darling_pcb.png
-rw-rw-r-- 1 xengi xengi  39K 2019-03-26 00:02:49 darling_schema.png
-rw-rw-r-- 1 xengi xengi    2 2019-03-26 00:44:55 fp-info-cache
-rw-rw-r-- 1 xengi xengi  125 2019-03-24 17:57:53 fp-lib-table
-rw-rw-r-- 1 xengi xengi 114K 2019-03-26 00:24:15 preview.png
-rw-rw-r-- 1 xengi xengi 480K 2019-03-26 00:41:05 preview.xcf
-rw-rw-r-- 1 xengi xengi 142K 2019-03-26 00:41:38 preview_002.png
$ gerbolyze vectorize bottom . out/ preview_002.png
Target layer file darling-B.SilkS.gbr
Outline layer file darling-Edge.Cuts.gbr
Parsing outline gerber
  * outline has offset (117.0, -130.5), size (24.0, 14.0)
Parsing target gerber
  * target layer has offset (124.238095, -129.816666), size (13.200000000000003, 12.307142999999996)
  * source image has size (1575, 919), going for scale 66dpmm
Combining keepout composite
  * outline
  * target layer
  * extra layer darling-B.Mask.gbr
Rendering keepout composite
Expanding keepout composite by 21
Thresholding source image
Padding source image
off (4, 2) end (1579, 921) img (1575, 919) tgt (1584, 924)
Masking source image
Calculating contour lines
    Extracting contours
Traceback (most recent call last):
  File "/home/xengi/.local/bin/gerbolyze", line 32, in <module>
    gerbolyze.process_gerbers(args.source, args.target, args.image, args.side, args.layer, args.debugdir)
  File "/home/xengi/.local/venvs/gerbolyze/lib/python3.7/site-packages/gerbolyze.py", line 342, in process_gerbers
    out = paste_image(tgrb, ogrb, source_img, [subtract], debugdir=debugdir, status_print=lambda *args: print(*args, flush=True))
  File "/home/xengi/.local/venvs/gerbolyze/lib/python3.7/site-packages/gerbolyze.py", line 208, in paste_image
    status_print=lambda *args: status_print('   ', *args))
  File "/home/xengi/.local/venvs/gerbolyze/lib/python3.7/site-packages/gerbolyze.py", line 231, in plot_contours
    img_cont_out, contours, hierarchy = cv2.findContours(img, cv2.RETR_TREE, cv2.CHAIN_APPROX_TC89_KCOS)
ValueError: not enough values to unpack (expected 3, got 2)
```

jaseg, 2019-03-26:
Fixed in v0.1.10 on the release branch. The issue here is that both OpenCV 3.*!=3.4 and 4.0 are API-incompatible with 3.4.

jaseg, 2019-03-26:
Also, this fix is on pypi already. Thanks for the report ;)

closed 2019-03-26 by jaseg