jaseg/gerbolyze #3: very cool!

badgeek, 2019-06-15:
thanks! this is what i always want to do with svg2sz but have no time and clue, im looking a way to integrate to svg2sz...

jaseg, 2019-06-16:
Thanks :)

The 'cool' way to integrate this into svg2sz would be to use it for bitmap graphics embedded into SVG files. The core logic is based on opencv and is pretty simple so it shouldn't be too much trouble to port it to C/C++ and emit KiCAD primitives instead of Gerber primitives. The main issue I see there is that KiCAD might not like that many primitives at once and just get slow, but I haven't tried.

If you need any help understanding the code feel free to ask.

neinseg, 2021-02-17:
@badgeek have a look at version 2, now on the main branch: Gerbolyze has now learned to both read SVG with almost full spec support and it can now generate kicad_mod files from arbitrary svg input by doing some funky (and a bit slow) computer geometry. The old code for tracing bitmaps is still there, but I have added some new fancy halftone processors for grayscale images as well.

Feel free to yoink anything you need :) The main geometry core and SVG parsing foo is in the [svg-flatten dir](https://github.com/jaseg/gerbolyze/tree/main/svg-flatten/src)

SVG example:
[image]

Halftone processing example (left is output zoomed out, right is zoomed in):
[image]

closed 2021-02-17 by neinseg

bakroistvan, 2021-03-01:
really nice