jaseg/gerbolyze #6: gerbolyze only fills coutouts

Jan--Henrik, 2020-03-02:
Hey, I ran into a problem. My board has a few cutouts and `gerbolyze vectorize` rendered my image only onto/into the cutouts, is there a quick fix for this problem?

Thank you in advance, gerbolyze is an amazing tool! :)

jaseg, 2020-03-02:
My first guess would be that maybe your outline is not closed? gerbolyze uses a [very primitve algorithm](https://github.com/jaseg/gerbolyze/blob/master/gerbolyze.py#L60) for figuring out what's outside and what's inside your board.

If you email me your gerbers I can take a closer look.

I love your otter-iron and otter-pill btw ;)

Jan--Henrik, 2020-03-02:
Thanks for the fast answer, I emailed you the gerbers! :)

jaseg, 2020-03-02:
There's a bug in the upstream gerber parser that makes it miscalculate image bounding box size. It's simply taking the min/max x/y coordinates of primitive endpoints instead of using the bounding box, and that gives the wrong result for the protruding arc on the bottom-center of your gerbers. That leads to gerbolyze cutting of part of the outline, creating a hole. I have pushed a quick workaround to master and I'll open an issue upstream.

It should work now, but you'll have to re-render and resize/realign your source image since from gerbolyzes point of view your board's size is now different.

Jan--Henrik, 2020-03-02:
Works, thanks alot! :)

closed 2020-03-02 by Jan--Henrik