jaseg/gerbolyze #8: Use library other than cairosvg which cannot easily be used in Windows.

pmankow, 2020-03-29:
Error "OSError: dlopen() failed to load a library: cairo / cairo-2 / cairo-gobject-2"

Cairo is not a Python library but part of GTK+

MarinMikael, 2020-03-31:
Hello,
I do not remember the exact details, but I use Cairo provided by Conda for pcb-tools (de library behind gerbolyze) and it works well. 
https://anaconda.org/anaconda/cairo

Good luck.

neinseg, 2021-08-18:
Both Cairo and OpenCV dependencies have been removed now. Cairo's path flattening was pants anyway, so I replaced it with part of antigrain graphics. I have re-written the relevant parts of OpenCV (~500 LoC), which now live in their own cpp file in-tree. All remaining dependencies are mini-libraries that I have either vendored (Clipper, AGG) or submodule'd. With this, the only remaining external dependency is libstdc++.

closed 2021-08-18 by neinseg