jaseg/gerbolyze #9: Possible output resolution error with render command

raoufahmed248, 2020-05-19:
Hey first of all thanks for making this amazing tool!
I am running into a possible issue with how the render command is outputting an image. I have a pcb project with dimensions of 91x37mm(3.57x1.46 inches). However when I run the gerbolyze render command on the gerber files (gerbolyze render top gerber_files output.png), I get a png file with a resolution of 96 pixels(W) by 235 pixels(H). 
However according to the default resolution setting (10px/6mils) I would be expecting a width of 2427 pixels (1456 mils / 6 mils * 10 pixels). I was able to get around this by setting the fab resolution to .238 which left me with a width of 2421 which was closer to what I was expecting. I'll be using the new output with custom fab resolution but I just wanted to let you know and was wondering if I am just doing something completely wrong.  I'll include the gerber folder and 2 outputs in-case you wanna look into it.

[test_files1.zip](https://github.com/jaseg/gerbolyze/files/4648363/test_files1.zip)

jaseg, 2020-05-19:
Hi,

thanks for your detailed bug report! I was able to reproduce the problem and have pushed a fix to master. The issue was that your input files are using inches as gerber units and gerbolyze expected millimeters. Now the codes auto-detects units from the outline layer.

Best, jaseg

raoufahmed248, 2020-05-19:
Thanks that fixed the issue!

closed 2020-05-19 by raoufahmed248