jaseg/gerbolyze #11: Problem with SVG image containing cutouts

plut0nium, 2021-02-16:
I am trying to use svg-flatten to convert SVG images to kicad footprints (silkscreen graphics).
My drawings contain multiple closed shapes with cutouts but these are rendered as plain filled areas, here are original image from inkscape and result in kicad:
[image]
[image]

Am I missing something or do I need specific preparation of the SVG file ?

The command line I use is:
`svg-flatten --format kicad --sexp-layer F.SilkS --sexp-mod-name tentacles_02 tentacles2.svg ../pcb/drawings.pretty/tentacles_02.kicad_mod`

Thank you very much 😀

neinseg, 2021-02-17:
Thank you for the report! Would you mind sharing the input SVG file with me at github@jaseg.de? This should work, so chances are you've found a bug.

neinseg, 2021-02-17:
Thank you for the test file. I have had a look and your problem should be fixed in v2.0.6, now on the main branch and on pypi.

The underlying issue was that your SVG for some reason contained only open paths and no closed paths at all. In SVG, an open path is one whose two ends are not connected, like the shape of the letter "C" vs. "O". In SVG, filling open subpaths like that is still well-defined behavior ([spec](https://www.w3.org/TR/SVG/painting.html#SpecifyingFillPaint)) even though here it likely is not what you want since when setting a stroke, things will look weird around the spot where the two ends of the path are.

Gerbolyze now handles this case according to spec and the resulting kicad_mod looks as you would expect. Please check it out, and feel free to re-open this issue or to open another issue should it not work as expected.

closed 2021-02-17 by neinseg