jaseg/gerbolyze #12: Loss of one vertex in each element when converting in KLayout

mx-yah, 2021-03-19:
Prepared Gerber from SVG 
svg-flatten --format gerber in.svg outfile.gbr

And imported outfile.gbr  in KLayout, in KLayout wiely that each element disappeared one vertex.
With gerbera files received in other ways such an error is not selected.

I did not find a way to invert svg

neinseg, 2021-03-19:
Can reproduce. Test input and output attached.
[test data issue 12.zip](https://github.com/jaseg/gerbolyze/files/6173314/test.data.issue.12.zip)

Klayout screenshot:
[image]

neinseg, 2021-03-19:
Okay, this is clearly a bug in KLayout.

gEDA gerbv, KiCAD gerbview **and Ucamco's reference gerber viewer** all produce correct output. I'll see whether I can raise an issue with KLayout's upstream.

[Ucamco's reference gerber viewer](https://gerber-viewer.ucamco.com/):
[image]

gEDA gerbv:
[image]

KiCAD gerbview:
[image]

neinseg, 2021-03-19:
Closing this, raised an issue with KLayout.
Thank you for the report!

closed 2021-03-19 by neinseg