jaseg/gerbolyze #13: inverting gerber

mx-yah, 2021-03-19:
Treatment of SVG from which black background and white letters freezes.
Tried
svg-flatten --format gerber outfile004_inv.svg outfile004_inv.gbr

I did not find the key to invert SVG when converting in gerber

neinseg, 2021-03-19:
Reproduced. Gerbolyze does not currently handle SVG background color at all and it does not allow flipping of either gerber polarity or the polarity of its SVG color -> Gerber polarity mapping. This means when you have an SVG document that has its background color set to black and has white elements on top, gerbolyze will render those as ```%LPD*%``` dark gerber primitives, but on an already "dark" gerber background. This means gerber viewers will produce no output.

As a work-around, in the source SVG you can add a black rectangle behind everything else in the document. See the screenshot below for an example.
[image]

And the result in gerbv:
[image]

Ideally, gerbolyze should export several switches to customize this behavior:
* Switchable SVG color interpretation: something like ```--clear-color/--dark-color``` but for the input not the output
* Handling of SVG background color. The sensible way to do this is probably to simply add a rect of the appropriate color below everything.

Gerbolyze's current behavior clearly is a bug since it violates gerbolyze's WYSIWYG goal.

neinseg, 2021-03-19:
Thank you for the report, btw!

neinseg, 2021-03-19:
The current main branch now supports an ```-f/--flip-gerber-polarity``` flag that will flip the gerber output's polarity. In your case, running ```svg-flatten --flip-gerber-polarity``` will produce the expected output.

I'll close this and open another issue for adding proper background handling, because doing that in a way that always works is a bit complicated.

closed 2021-03-19 by neinseg

mx-yah, 2021-03-24:
Hello!
Please advise settings to convert this file to Gerber without distortion

    суббота,

 
 

Closed #13.

mx-yah, 2021-03-24:
[positiv-negativ_send.svg.zip](https://github.com/jaseg/gerbolyze/files/6197917/positiv-negativ_send.svg.zip)