jaseg/gerbolyze #16: MacOS support?

theacodes, 2021-04-04:
Howdy, thank you for all of the work you've put into this! I would love to be able to use this on MacOS.

Since this could significantly make my life easier I'm happy to pay to see this feature - either via github sponsorship, ko-fi, paypal, or a donation to a non-profit of your choice.

Thanks again for this! 💜

jaseg, 2021-04-05:
Hey there! I recognize your name from reading your blog, I love your work ^^

I've done a bit of work, and gerbolyze now installs on macOS just fine.  It only took two very minor changes. I've added instructions [in the README](https://github.com/jaseg/gerbolyze#macos-via-homebrew) and uploaded v2.0.8 including the fixes to PyPI.

Note that this *does* require installing the dependencies via homebrew (which AFAICT is what all mac people use anyway??). The only alternative would be for me to statically link or otherwise package these dependencies with the python package on PyPI. I don't want to do that (yet) because the resulting package would be extremely large, and I'd still not be confident that it would run on different versions of macOS. At some point in the future I'm going to solve this issue by cutting down gerbolyze's dependencies. It only uses very small parts of the bulkier ones: cario, pango and opencv. Until then, please bear with me :)

Please tell me if it works for you.

theacodes, 2021-04-07:
Awesome, thank you so much! I'm gonna try this out this weekend and I'll let you know how it goes.

jaseg, 2022-06-17:
With gerbolyze 3.0 out, x86_64 MacOS is now supported with a simple 'pip install'. No system binaries are needed anymore, and everything is downloaded pre-built from PyPI.

aarch64 (M1/M2) MacOS will be supported as soon as the aarch64 support in wasmtime has trickled down to the wasmtime python package.

closed 2022-06-17 by jaseg