jaseg/gerbolyze #17: Super low poly icons

lerouxb, 2021-04-17:
I'm trying to get some icons onto my silkscreen, but no matter what I try it comes out super "low resolution", for lack of a better term. Specifically circles seem to become rotated squares.

For example, I tried various music note icons from various websites:

https://www.svgrepo.com/vectors/music/monocolor/
https://feathericons.com/?query=note
https://remixicon.com/

(any of those should do)

I put them on the silk layer in inkscape. I've tried embedding png images but that didn't get included in the silk layer at all. Had to draw some primitives onto the silk layer to make sure the process is working end to end. All I've gotten to show up on the gerber is if I drag in a svg file and choose to include it (first radio button. ie. not as an image tag).

[image]
[image]

This is just a test to get some screenshots. When I make the icons smaller (which would be my intention for how I'd want to use it in this case) the effect is even more pronounced. 

Is there something obvious I'm missing? A setting I have to change perhaps? Would be nice if the round parts looked a bit rounder.

neinseg, 2021-04-20:
I can reproduce this. svg-flatten has a "curve tolerance" setting that is supposed to control this, but it seems that these curves don't even hit the flattener as changing that setting does not have an effect. I will look into this in detail tonight.

neinseg, 2021-04-24:
I think I've fixed this issue. I've pushed the fix to the main branch. Please feel free to try it out.

For both `svg-flatten` and `gerbolyze`, the `--curve-tolerance` command-line argument now does what it says on the tin and sets the maximum tolerance for bezier flattening in mm. Gerbolyze will now happily generate micrometer-smooth curves in gerber files tens of megabytes in size ;)

The fix ended up being to completely rip out Cairo and replace its bezier flattener with part of Antigrain Graphics' one. A beneficial side effect of this is that it is a step towards Windows builds, since on Windows the Cairo dependency was an issue.

Unfortunately, while fixing this issue I have re-introduced a bug where clipper calculates the mask layer cutouts too small. This results in silkscreen being closer than intended to soldermask openings in some cases. As all board houses will remove any silk overlay on pads anyway this is a purely cosmetic issue, but I will hold off with a release of my fix until I have also fixed this problem.

[image]

closed 2021-08-18 by neinseg