jaseg/gerbolyze #19: https://gerbolyze.jaseg.net/ only gives `Internal Server Error`

amadejhp, 2021-04-21:
I tried using the website for this code, but it only gives me `Internal Server Error` when trying to open/download files. Is it possible to get it fixed? I didn't find any contact info on it so I assumed this would be the place to ask.

amadejhp, 2021-04-21:
If that isn't possible, could there be a simple tutorial, for someone who never used github before, how to install it on Windows 10?

neinseg, 2021-04-22:
I'm sorry, the website is currently broken since my cheap virtual server does not have enough RAM to run it. Since finishing v1.0 I didn't take the time yet to fix it. I'll try to take some time next weekend to either fix the thing or to at least put up a page explaining things.

I'm using gerbolyze on Windows myself. The easiest way to do that is to install any supported Linux distro (Debian/Ubuntu/Fedora/Arch) through Windows 10's built-in "WSL" Linux emulation layer. Microsoft's official docs on that are [here](https://docs.microsoft.com/en-us/windows/wsl/install-win10). Once you have got WSL up and running, you can use the WSL terminal to follow the steps that the README lists [for the linux distribution you chose](https://github.com/jaseg/gerbolyze#quick-start-installation) in WSL.

amadejhp, 2021-04-22:
Is there any simpler way to get this? Like an exe or at least just extractable zip file? or a way to run the website locally? I'm trying to install WSL but no luck. Tried getting windows insider version but now it refuses to open update settings... Tbh I'd donate if you can patch together a simpler to use version. I just want to make PCB art xD

neinseg, 2021-04-23:
Hey there,

I'm glad you got it working via WSL!

**Windows version:**
I'm currently planning to get the website back up and running, and I think that in the short term this will be the best solution for a windows-only workflow. Currently, cross-compiling for windows is tricky since gerbolyze requires some complex dependencies (cairo and OpenCV). I'm also working on that in parallel, but it will take some time to get that sorted out. I'll try to find some time to fix at least the website this weekend.

**Usage:**
The Gerbolyze workflow uses one SVG file per board side. To put designs on both sides, it is easiest to call it twice, once for each side. Each "side" SVG contains copper, mask, paste and silk layers for this side.

1. In step one, you convert the gerbers that come out of your CAD tool into an Inkscape-compatible SVG "template" using ``gerbolyze template``. You select the side with `--top` or `--bottom`. The template has Inkscape layers for all Gerber layers on that side and contains a preview of the board layout and the board outline for positioning. Note that currently, gerbolyze can't output to the board outline layer in a way that PCB fabs will like.
2. You then open the template SVG in Inkscape and place your artwork on the appropriate layers. Each Inkscape layer modifies one Geber layer. Make sure you select the right one in the bottom of Inkscape's main window. Other SVG tools should also work, but might not recognize Inkscape's layers (which are actually overlapping empty SVG groups). After editing, save the edited SVG somewhere. It's best to use a new file name, otherwise it's easy to accidentially overwrite the file when calling `gerbolyze template` a second time.
3. Finally, you call ``gerbolyze paste`` with the original gerbers and your edited template to produce the output gerbers.

For example to generate a top-side template, in the WSL shell you might run:

```shell
$ gerbolyze template --top /mnt/c/Users/jaseg/Documents/8seg/pcb/top_design.svg /mnt/c/Users/jaseg/Documents/8seg/pcb/gerber
```

This will read the gerber files in the Windows ` C:\Users\jaseg\Documents\8seg\pcb\gerber` folder and create `C:\Users\jaseg\Documents\8seg\pcb\top_design.svg` from them. In WSL, `/mnt/c` points to your Windows `C:\` drive ([documentation](https://docs.microsoft.com/en-us/windows/wsl/faq#how-do-i-access-my-c--drive-)).

After editing the SVG with inkscape and saving it as `top_design_edited.svg`, this would be the second step:

```shell
$ gerbolyze paste --top /mnt/c/Users/jaseg/Documents/8seg/pcb/top_design_edited.svg /mnt/c/Users/jaseg/Documents/8seg/pcb/gerber /mnt/c/Users/jaseg/Documents/8seg/pcb/gerber_with_design
```

The syntax is `gerbolyze paste --top [SVG] [input gerber folder] [output gerber folder]`

This will read the gerber files from the `gerber` folder, add the design from the SVG to the corresponding layers and write the modified gerber files to the `gerber_with_design` folder. If this folder does not exist yet, it will create it.

amadejhp, 2021-04-23:
Thank you! That makes more sense now.

Almost there, I figured out I need to rename the .gbr files to .gtl and .gtb and .gko in order for script to detect them, and I successfully got a template saved and opened in Inkscape :)

Now it stops at the paste command, I'm guessing `Error: Cannot find usvg. Is it installed and in $PATH?` could be the reason?

[image]

One more question, where do I add the `--vectorizer hex-grid` to get that pattern from an image?

neinseg, 2021-04-24:
**usvg not found issue**

From the [Ubuntu quickstart guide](https://github.com/jaseg/gerbolyze#ubuntu), could you please run the line `cargo install usvg` again and post the output of the command here? If this line ran correctly, gerbolyze should be able to find usvg.

**Gerber file names**

I actually try to have gerbolyze guess these file names correctly. Would you mind telling me what CAD tool you're using so I can add its file name endings to Gerbolyze's match list?

**vectorizer settings**

The `--vectorizer` argument goes like this: `gerbolyze paste --vectorizer hex-grid`

See `gerbolyze paste --help` and `svg-flatten --help` for details.

**Stand-alone vectorizer usage**

All the SVG magic behind gerbolyze is implemented in the `svg-flatten` tool that gets installed next to gerbolyze. You can find its usage instructions [in the README](https://github.com/jaseg/gerbolyze#id2) or by running `svg-flatten --help`. Gerbolyze calls `svg-flatten` once for every layer, but you can also call it yourself. When you call it yourself, you can pass it not only SVG files but also PNG files. It can output either gerber or KiCAD data or SVG, so you can use it to convert bitmaps to SVGs. Just be careful, as at high resolutions it will happily create an SVG with thousands of little paths that may hang some SVG editors ;)

amadejhp, 2021-04-24:
**Usvg**
[image]
Tested it and it looks like it works now, Thank you! This is amazing software, tricky to set-up but the results are worth it.

Vectorizer seems to run without trouble, but when I try to open the svg it says it is empty?
[image]
[image]
I could really use this feature for another project of mine, posters and such.

neinseg, 2021-04-24:
**Outlines**
Outlines do in fact not work yet in Gerbolyze. The reason for that is that Gerbolyze processes all strokes to Gerber "region" areas instead of normal Gerber movement commands. For now, I would recommend using either [KiCAD StepUp](https://github.com/easyw/kicadStepUpMod) via FreeCAD or KiCAD's DXF import with a DXF file exported from e.g. Inkscape. I will try to find some time this or next weekend to add proper outline support to Gerbolyze.

**Empty SVGs**
The command line in your screenshot is missing `-o svg`. Without it, `svg-flatten` produces Gerber instead of SVG. Something like this should work: `svg-flatten -o svg -s 42.3x42.3 --vectorizer hex-grid in.png out.svg`

amadejhp, 2021-04-25:
Oh okay. If you do manage to get them working, that would be stellar! Do let me know when/if you get it working. I don't really have any experience making working PCBs, I'm more of an artist type heh.

**Svg**
Oops, thanks, at least it's an easy fix

And seriously man, keep up the great work!

amadejhp, 2021-04-25:
Btw, I'm having issues with JLCPCB if you have any experience with them? Even a hand-drawn outline does not get detected on their site for some reason

neinseg, 2021-04-26:
I have done several boards with them, including some made with gerbolyze. Their online gerber viewer is pretty bad and I think it actually renders some things incorrectly. I mostly use [gEDA gerbv](http://gerbv.geda-project.org/) for validation. I can also recommend [Ucamco's reference Gerber viewer](https://gerber-viewer.ucamco.com/). Ucamco are the ones who *make* the Gerber standard, so what their reference viewer shows is what must per standard come out.

If something is looking ok in both gerbv and Ucamco's reference viewer, then it is likely to come out ok. I have had a few designs that looked ok in both but were rendered wrong by JLC's online viewer. When in doubt about some particular feature, at the bottom of the order screen you can also enter a comment to their CAM engineers asking them to check.

My most frequent issue with manufacturing gerbolyze-made is that they sometimes have small specks that are too small for the fab. In these cases, the fabs are usually ok if I tell them "please please" and accept responsibility for manufacturing defects caused by this design rule violation.

amadejhp, 2021-04-26:
Gotcha, thank you, I'll try them.
Also, I sent an outline to their support and they said it will work fine and they're sorry their viewer is broken.

Yeah, it's expected that some spots might be too small, but it shouldn't be that bad without them.

Zepthoy, 2021-05-10:
Hi, I have also used WLS. I have generated the .svg file with ` $ gerbolyze template --bottom 
 /mnt/c/Users/Zepthoy/Documents/bottom_design.svg /mnt/c/Users/Zepthoy/Documents/gerber ` 
 
 
[image]

 
 and I have used `svg-flatten -o svg -s 42.3x42.3 --vectorizer hex-grid in.png out.svg` to vectorize my PNG. 

[image]

But when I use gerbolyze paste command, no layer is modified and the gerber generated does not contain art. 

[image]

I don't know what the problem could be, I have also tried to paste the art directly into Inkscape.

jaseg, 2022-07-11:
The webapp is online again at https://tsugi.kokoroyukuma.de/gerboweb

arduino12, 2022-09-20:
**Hi,**

The webpage won't open :(

If you can fix it please also update the README:
"Try gerbolyze online at https://dyna.kokoroyukuma.de/gerboweb"

Hope to visit the webpage as soon as it works,
**Thanks!**

jaseg, 2022-10-12:
The webpage is online once more at https://dyna.kokoroyukuma.de/gerboweb/

arduino12, 2022-10-14:
I tried uploading the [testdata](https://github.com/jaseg/gerbolyze/tree/main/testdata) files (zipped all gerbers) to the [webpage](https://dyna.kokoroyukuma.de/gerboweb/)-
But after clicking "Submit" nothing happens.. I don't see any error message - only the page is refreshed.

Am I doing it wrong or is it a server error?