jaseg/gerbolyze #20: Using paste command / Problem with gerbv

FPittelkow, 2021-05-22:
Hi,

when I try to use  `gerbolyze paste` I get an error. It seems like there is a problem with gerbv. But I'm not sure, if I did something wrong or if something else is broken.
I'm on macOs 11.3.1

```
subprocess.CalledProcessError: Command '['gerbv', '-x', 'svg', '--border=0', '--origin=4.972441x-5.032677', '--window_inch=3.103150x1.131890', '--foreground=#ffffff', '-o', '/var/folders/dd/fmghlwq1677bxlf1n9f_qjwm0000gn/T/tmpu2mco5z3/dilate-tmp.svg', 'oscger/OSC-F_Mask.gbr']' returned non-zero exit status 1.
```

neinseg, 2021-05-22:
Hey there!

Could you please post the output of running the command from the error message manually in your terminal? Here's the command:

`gerbv -x svg --border=0 --origin=4.972441x-5.032677 --window_inch=3.103150x1.131890 --foreground=#ffffff -o /tmp/test.svg oscger/OSC-F_Mask.gbr`

FPittelkow, 2021-05-23:
Hi,

the output of running the command:

`Specified window size is out of bounds.`

neinseg, 2021-05-24:
That is interesting. The error is probably with gerbolyze in that case. Would you mind sharing your gerbers with me through this issue or the email listed on my account page?

olivaq, 2021-09-25:
I ran into the same error message, for my part it turned out it was the number locale settings (norwegian) on my system that specified comma as decimal point. When gerbv comes along parsing the floats from gerbolyze it ignores the period as it's just a thousand seperator in my locale, resulting in big numbers that's out of bounds.

Changing to US locale worked.

There is a 'n' number format option that follows locale, but it didn't seem python picked up locale from the system automagically so I had to set it to make that work..

jaseg, 2022-06-17:
This is fixed in gerbolyze 3.0. Gerbolyze now installs all dependencies pre-built via PyPI, and does not require any system dependencies anymore.