jaseg/gerbolyze #21: No Change using --scale or --usvg-dpi

alorman, 2021-08-11:
Hi;
This is an incredible toolbox, well done on making it!

I've been experiencing an issue related to illustrator exporting SVG that are scaled wrong. This has to do with the `viewbox` field in the SVG. There are numerous issues in online forums about this within illustrator.

I thought I might be able to account for this (scaling by 0.66) in command line with Gerbolyze. 
I've invoked `svg-flatten --format svg --scale 0.66 /home/path/to/svg.svg /home/path/to/out.svg`
I've also tried 
`svg-flatten --format kicad --sexp-mod-name TestModule --scale 0.66 /home/path/to/svg,svg /home/path/to/out.kicad_mod`

I've also tried to address this with the `--usvg-dpi` flag, but also have seen no effect.
I've tried `svg-flatten --format kicad --sexp-mod-name TestModule --usvg-dpi 72 home/path/to/svg.svg /home/path/to/out.kicad_mod`

I've attached a troublesome SVG (hilarious that Github doesn't support SVG uploads) to this post if anyone wants to experiment. It should be 53.3 x 32.3 mm. It seems to always come into KiCad as 151 x 92 mm
[2_nostroke_clipping_mask_noresp.zip](https://github.com/jaseg/gerbolyze/files/6965449/2_nostroke_clipping_mask_noresp.zip)

neinseg, 2021-08-17:
Thank you for the report.

The issue was that svg-flatten expected the document unit to be mm, and did not properly scale document contents otherwise. I have fixed this issue on master.  svg-flatten now correctly handles both `viewBox` and `width` / `height` for arbitrary document units.. I will release a new version to PyPI some time next week.

The `--scale` argument is only meant for gerber output atm, I will fix the documentation on that.

closed 2021-08-17 by neinseg

alorman, 2021-08-17:
@neinseg  Awesome! Thanks!!