jaseg/gerbolyze #22: Possible Windows (or alternately web) version?

boysrbackintown, 2022-01-29:
Hi!

I've been following the project for a while now and I love the results you are getting. Dithering rocks.

I see that there is a macOS build via homebrew. Is there an easy way to run this in Windows without resorting to a docker container or something similar? A way to build this in windows would be great, or something akin to homebrew, or even just a little portable app. I know a webUI version is far more complicated as well but I've seen a lot of projects in this space also use those so that might be looking into. 

Thank you so much for your work so far! The results are gorgeous.

boysrbackintown, 2022-02-04:
That's wonderful to hear. I'm really excited to see what you have in store. Can't wait.

Best,
Chris

jaseg, 2022-06-17:
gerbolyze 3.0 is out now, and can be installed from PyPI on any platform using pip. It does not require any host utilities anymore, so running it on Windows should be as easy as ```pip install gerbolyze```.

closed 2022-06-17 by jaseg