jaseg/gerbolyze #23: FileNotFoundError: [Errno 2] No such file or directory: 'gerbv'

chrpsh, 2022-01-31:
hello !

i'm trying to convert my edited svg template back to gerbera, but got an error:
`FileNotFoundError: [Errno 2] No such file or directory: 'gerbv'`

full log:
```
$ gerbolyze paste --top /Users/chrpsh/Downloads/EMIMic_v2/EMIMic_v2.preview-top.svg /Users/chrpsh/Downloads/EMIMic_v2 /Users/chrpsh/Downloads/EMIMic_v2/exp_tst

#########################################
processing  top input file  /Users/chrpsh/Downloads/EMIMic_v2/EMIMic_v2.preview-top.svg
#########################################

loading silk layer from: /Users/chrpsh/Downloads/EMIMic_v2/EMIMic_v2.GTO
loading mask layer from: /Users/chrpsh/Downloads/EMIMic_v2/EMIMic_v2.GTS
loading copper layer from: /Users/chrpsh/Downloads/EMIMic_v2/EMIMic_v2.GTL
loading outline layer from: /Users/chrpsh/Downloads/EMIMic_v2/EMIMic_v2.GM1
loading drill layer from: /Users/chrpsh/Downloads/EMIMic_v2/EMIMic_v2.TXT
loaded layers: ['silk', 'mask', 'copper', 'outline', 'drill']
bounds: ((0.0, 2.1063), (0.0, 3.7795))

-----------------------------------------
processing side top layer silk
-----------------------------------------

rendering layer silk
svg-flatten args: --format gerber --precision 6 --trace-space 0.1 --only-groups g-silk /Users/chrpsh/Downloads/EMIMic_v2/EMIMic_v2.preview-top.svg /var/folders/bp/j7_pnwr93zb0352wfgrjmvk00000gn/T/tmpb7iolkkb/overlay-top-silk.gbr
copying svg input into temp svg
calling usvg on "/var/tmp/tmp.0.SauvhX.svg" and "/var/tmp/tmp.1.Xr55dy.svg"
parsed 0 vectorizers
Loading temporary file "/var/tmp/tmp.1.Xr55dy.svg"
loaded viewbox: 0, 0, 2.1063, 3.7795
document viewbox clip: bbox={0, 0} - {21063000, 37795000}
Forwarding layer name to sink: "g-silk"
used svg-flatten: svg-flatten
compositing
processing dilation mask 0.1
dilating mask by 0.1
dilation cmd: gerbv -x svg --border=0 --origin=0.000000x0.000000 --window_inch=0.082925x0.148799 --foreground=#ffffff -o /var/folders/bp/j7_pnwr93zb0352wfgrjmvk00000gn/T/tmpb7iolkkb/dilate-tmp.svg /Users/chrpsh/Downloads/EMIMic_v2/EMIMic_v2.GTS
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.10/bin/gerbolyze", line 33, in <module>
    sys.exit(load_entry_point('gerbolyze==2.1.1', 'console_scripts', 'gerbolyze')())
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/gerbolyze.py", line 181, in paste
    dilated = do_dilate(d_layer, amount)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/gerbolyze.py", line 139, in do_dilate
    dilate_gerber(layers, layer, amount, bbox, tmpdir, outfile, units, curve_tolerance)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/gerbolyze.py", line 599, in dilate_gerber
    subprocess.run(cmd, check=True, stdout=subprocess.DEVNULL, stderr=subprocess.DEVNULL)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/subprocess.py", line 966, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/subprocess.py", line 1842, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'gerbv'
```

how can i fix this? (tried to use `sudo`, same output)
thank you

jaseg, 2022-02-01:
Hi there,

gerbolyze needs gerbv installed. gerbv is an open source gerber viewer program. Your distro will likely have a "gerbv" package that you can install using apt/pacman/dnf/etc.

chrpsh, 2022-02-04:
is it a problem with my svg or something?

jaseg, 2022-06-17:
With gerbolyze 3.0 out, this issue is resolved. Gerbolyze now depends only on other python packages, and needs no host tools anymore.

closed 2022-06-17 by jaseg