jaseg/gerbolyze #27: Hi there, thanks for the package. Is there any way we can integrate this into a python file?? any resources that we can refer? I want to convert svg file into gerber and show them in a website

surendhar-palanisamy, 2022-07-20:


jaseg, 2022-07-23:
I'm not sure I understand your question, but here's some answers anyway:

If you want to automatically call gerbolyze from a python script, the easiest way to do that is through the command-line interface (i.e. something like `subprocess.run(['gerbolyze', ...`). This is because gerbolyze does not have an API since there is not really a use case for one. If this is what you want to do, please be aware that gerbolyze is under an AGPL license, so you will need permission to embed it in a website that is not itself open source.

If you want to use gerbolyze on the web, there is the demo webpage at https://dyna.kokoroyukuma.de/gerboweb/ that you can use to run gerbolyze without installing it.

If you want to show the gerber output from gerbolyze on a webpage, gerbolyze alone is not enough. One way to embed gerber in a web page is to convert it to SVG and embed that as an <img>. You can convert gerber to SVG using either `gerbonara render` (part of gerbolyze), or you can use e.g. gerbv, which has an SVG output option. Another option would be to embed a Javascript gerber viewer such as tracespace in the website.

closed 2022-07-23 by jaseg

surendhar-palanisamy, 2022-07-24:
Thanks for the reply :)