jaseg/gerbolyze #28: PermissionError: [Errno 13] on Gerbolyze Paste

ronanarm, 2022-07-22:
When using Gerbolyze paste, the program throws an error after the Overlay Mechanical Outline layer.

In this instance, KicadGerberOUT is the original set of gerber files, input.svg is an inkspace SVG made with the template from gerbolyze template, and KicadGerberIN is an empty output directory.

```
gerbolyze paste KicadGerberOUT input.svg KicadGerberIN
```

Output error and traceback:
```
Traceback (most recent call last):
  File "/usr/local/bin/gerbolyze", line 4, in <module>
    __import__('pkg_resources').run_script('gerbolyze==3.0.11', 'gerbolyze')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 667, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1463, in run_script
    exec(code, namespace, namespace)
  File "/usr/local/lib/python3.8/dist-packages/gerbolyze-3.0.11-py3.8.egg/EGG-INFO/scripts/gerbolyze", line 5, in <module>
    gerbolyze.cli()
  File "/home/rorofat/.local/lib/python3.8/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/rorofat/.local/lib/python3.8/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/rorofat/.local/lib/python3.8/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/rorofat/.local/lib/python3.8/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/rorofat/.local/lib/python3.8/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/gerbolyze-3.0.11-py3.8.egg/gerbolyze/__init__.py", line 92, in paste
    stack.save_to_directory(output_gerbers)
  File "/usr/local/lib/python3.8/dist-packages/gerbonara-0.11.1-py3.8.egg/gerbonara/layers.py", line 426, in save_to_directory
    layer.save(out)
  File "/usr/local/lib/python3.8/dist-packages/gerbonara-0.11.1-py3.8.egg/gerbonara/cam.py", line 419, in save
    shutil.copy(self.original_path, filename)
  File "/usr/lib/python3.8/shutil.py", line 418, in copy
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/lib/python3.8/shutil.py", line 264, in copyfile
    with open(src, 'rb') as fsrc, open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permission denied: 'KicadGerberIN/PCB Json-F_Cu.gbr'
```

ronanarm, 2022-07-22:
So it appears that the output directory cannot exist prior to running the gerbolyze paste command. Hope this helps someone else!

closed 2022-07-22 by ronanarm

jaseg, 2022-07-23:
Hi there,
thank you for the update! Your problem may have been caused by permissions being weird on your output directory and not allowing gerbolyze to create or delete files. By deleting and re-creating the output directory, owner and permissions get reset and that may have fixed the problem for you.

Please note that in your original post, the input and output gerber dir arguments look like they might be flipped: The order should be `gerbolyze paste INPUT SVG OUTPUT`.