jaseg/gerbolyze #30: Error when trying to create template

regicidalplutophage, 2022-10-23:
```
  File "/root/.local/lib/python3.10/site-packages/gerbonara/graphic_primitives.py", line 217, in flip
    cx=(self.x + self.cx) - self.x2, cy=(self.y + self.cy) - self.y2, clockwise=not self.clockwise)
AttributeError: 'Arc' object has no attribute 'x'. Did you mean: 'cx'?
root@home-pc:~#
```
After poking around in graphic_primitives.py I did manage to get and SVG output but it didn't seem to be right.

regicidalplutophage, 2022-10-27:
Solved this by downgrading to gerbonara 0.11.1 and installing all the binaries.
Latter was probably required because of whacky initial installation caused by me tring different versions of python with pyenv.

Is current version of gerbolyze not ready for gerbonara 0.11.2?
If so, maybe it should be reflected in installation guide.

I'm still not sure if everything is working as it's intended:
when doing `gerbolyze template --bottom` the svg is flipped vertically and had to be unflipped for `gerbolyze paste` output to match it, which defeats the purpose of having separate top and bottom templates.
Should I create a separate issue for it?

enbeec, 2024-02-22:
How did you do the downgrade? I'm at a bit of a loss.