jaseg/gerbolyze #31: better instructions how to use this would be appreciated.

stefanputz, 2023-02-24:
I have a really hard time to get this running, help is appreciated.
If I run "gerbolyze convert test.svg test.gbr" on my machine (WSL linux on Windows 11) I get the following error which I am not sure what the problem is. I assume that the test.svg in the repository good, but I am not even sure about that?

used usvg: wasi-usvg
Info: Skipping usvg
used svg_flatten_wasi python package
Traceback (most recent call last):
  File "/home/stefan/.local/bin/gerbolyze", line 5, in <module>
    gerbolyze.cli()
  File "/home/stefan/anaconda3/lib/python3.9/site-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/home/stefan/anaconda3/lib/python3.9/site-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/home/stefan/anaconda3/lib/python3.9/site-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/stefan/anaconda3/lib/python3.9/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/stefan/anaconda3/lib/python3.9/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/home/stefan/.local/lib/python3.9/site-packages/gerbolyze/__init__.py", line 265, in convert
    bounds = stack.board_bounds()
  File "/home/stefan/anaconda3/lib/python3.9/site-packages/gerbonara/layers.py", line 516, in board_bounds
    if self.outline:
  File "/home/stefan/anaconda3/lib/python3.9/site-packages/gerbonara/layers.py", line 637, in outline
    return self['mechanical outline']
  File "/home/stefan/anaconda3/lib/python3.9/site-packages/gerbonara/layers.py", line 605, in __getitem__
    return self.graphic_layers[(side, use)]
KeyError: ('mechanical', 'outline')

regicidalplutophage, 2023-03-15:
Sadly this project seems to be abandoned. What made it work for me was downgrading gerbonara to 0.11.1

regicidalplutophage, 2023-03-15:
I lied. There are some recent commits on gitlab.

jaseg, 2023-03-18:
It's not abandoned, I was just busy IRL recently and haven't had time yet to brong gerbolyze's CI up to date with my recent update to gerbonara.

stefanputz, 2023-03-18:
> Sadly this project seems to be abandoned. What made it work for me was downgrading gerbonara to 0.11.1

I tried that but did not solve the issue, maybe I made some mistakes there

> It's not abandoned, I was just busy IRL recently and haven't had time yet to bring gerbolyze's CI up to date with my recent update to gerbonara.

First of all thank you for sharing this project and also replying!

It would be awesome if you could include a notebook with some examples which users can open and just run.
In the meantime I have come up with a solution myself using gerber-writer but I would like to compare it to your much more sophisticated stuff.