jaseg/gerbolyze #32: PIP install fails with "No matching distribution found for svg-flatten-wasi (from gerbolyze)"

puterboy, 2023-04-18:
I am trying to install gerbolyze in python3 venv (running under Ubuntu 18.04)

I get the following error:
```
$  python3 -m pip install --user gerbolyze
Collecting gerbolyze
  Using cached https://files.pythonhosted.org/packages/3f/2e/231077f2f8731e0725427aace057778e6aff5e2b11f3c7a805f81c1e2e22/gerbolyze-3.1.5-py3-none-any.whl
Collecting svg-flatten-wasi (from gerbolyze)
  Could not find a version that satisfies the requirement svg-flatten-wasi (from gerbolyze) (from versions: )
No matching distribution found for svg-flatten-wasi (from gerbolyze)```

jaseg, 2023-04-18:
svg-flatten-wasi requires python 3.7 and upwards. Gerbolyze also depends on gerbonara, which now requires python 3.10 and up. Ubuntu 18.04 currently runs python 3.6, which is too old to run gerbolyze or its dependencies. If you want to try it anyway, the only idea I have would be to try Anaconda, which AFAIK has built-in python version management.

Since gerbolyze is a hobby project, I don't have the time to support pythons older than two years or so, since that would require a lot of time creating workarounds and polyfills for missing features.

puterboy, 2023-04-19:
Completely understand the "hobby" part :)
Would be interesting if this could be made to run in a Docker envelope though I don't really have much if any experience with Docker myself...

puterboy, 2023-04-19:
I seem to have gotten it to run under docker. Was not to hard.
Here is what I did in case others want to replicate:
```docker pull ubuntu:22.04
docker run -v <local-gerber-dir>:<docker-gerber-dir> --name gerbolyze -it ubuntu:22.04 bash
    apt update
    apt install -y python3 python3-pip python3-venv
     mkdir /gerbolyze
     cd /gerbolyze
     python3 -m venv venv
     source venv/bin/activate
     pip install gerbolyze beutifulsoup4 resvg-wasi
```