jaseg/gerbolyze #35: SystemError: resvg executable not found

puterboy, 2023-04-19:
Based on the documentation it seems that one doesn't need to install 'resvg' unless one wants to speed up execution.

However, running 'gerbolyze template', I get the following error:
```Traceback (most recent call last):
  File "/mydata/gerbolyze/venv/bin/gerbolyze", line 5, in <module>
    gerbolyze.cli()
  File "/mydata/gerbolyze/venv/lib/python3.10/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/mydata/gerbolyze/venv/lib/python3.10/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/mydata/gerbolyze/venv/lib/python3.10/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/mydata/gerbolyze/venv/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/mydata/gerbolyze/venv/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/mydata/gerbolyze/venv/lib/python3.10/site-packages/gerbolyze/__init__.py", line 181, in template
    run_cargo_command('resvg', temp_svg.name, temp_png.name, dpi=f'{raster_dpi:.0f}')
  File "/mydata/gerbolyze/venv/lib/python3.10/site-packages/gerbolyze/__init__.py", line 432, in run_cargo_command
    raise SystemError(f'{binary} executable not found')
SystemError: resvg executable not found```

Note: I also don't have 'rust' or 'cargo' installed -- are they prerequisites?  I don't see them mentioned in the docs except in the seemingly optional section of speeding up with native binaries.

puterboy, 2023-04-19:
OK. I fixed this by manually running:
`python -m pip install resvg-wasi`

jaseg, 2023-04-19:
I've updated the README. Thanks!

closed 2023-04-19 by jaseg