jaseg/gerbolyze #38: How do you use this with EasyEDA Gerber files

puterboy, 2023-04-19:
So, after finally getting gerbalyze template & paste to install and run in an Ubuntu Docker python3.10-pip environment (both using a full 'git' install and separately just using 'pip install'), I loaded the output files in several different Gerber viewers but the image I added in Inkscape didn't show up in the newly created Gerber file.

My steps were as follows:

1. Export Gerber file from EasyEDA
2. Run `gerbalyze template --bottom easyeda.zip output.svg` - No errors though a couple of warnings (see below)
3. Open 'output.svg' in Inkskape  and add/resize/move my image (a dithered high res B&W photo) to the bottom silkscreen level (called 'bottom-silk'). Save the image to output-image.svg
4. Run `gerbalyze paste easyeda.zip output-image.svg easyeda-image.zip` - No errors though a couple of warnings (see below)
5. Load 'easyeda-image.zip' into a Gerber viewer

The resulting top and bottom board views looked the *same* as the original -- i.e., no image added.
When I looked inside and compared the new and original zip files, I found that

- Some layers in the original Gerber file were deleted (e.g., Gerber_TopAssemblyLayer.GTA, Gerber_DocumentLayer.GDL, Gerber_DrillDrawingLayer.GDD, How-to-order-PCB.txt -- presumably at least the last one is not really relevant)
- The individual file sizes were all slightly *smaller* -- hopefully, no essential data was lost
- No evidence of an image being added (as all files were smaller than their original counterparts)

Specifically, the files before looked like:
```Archive:  Gerber_PCB4_2023-04-18.zip
  Length      Date    Time    Name
---------  ---------- -----   ----
    12282  2023-04-18 21:47   Gerber_TopLayer.GTL
    34796  2023-04-18 21:47   Gerber_BottomLayer.GBL
   560729  2023-04-18 21:47   Gerber_TopSilkscreenLayer.GTO
    18684  2023-04-18 21:47   Gerber_BottomSilkscreenLayer.GBO
     3696  2023-04-18 21:47   Gerber_TopSolderMaskLayer.GTS
     2102  2023-04-18 21:47   Gerber_BottomSolderMaskLayer.GBS
     9819  2023-04-18 21:47   Gerber_TopPasteMaskLayer.GTP
      544  2023-04-18 21:47   Gerber_TopAssemblyLayer.GTA
      449  2023-04-18 21:47   Gerber_BoardOutlineLayer.GKO
     4293  2023-04-18 21:47   Gerber_DocumentLayer.GDL
    61079  2023-04-18 21:47   Gerber_DrillDrawingLayer.GDD
     1516  2023-04-18 21:47   Drill_PTH_Through.DRL
      700  2023-04-18 21:47   Drill_PTH_Through_Via.DRL
       76  2023-04-18 21:47   How-to-order-PCB.txt
---------                     -------
   710765                     14 files```
And after looked like:
```Archive:  Gerber_PCB4_2023-04-18-new.zip
  Length      Date    Time    Name
---------  ---------- -----   ----
    11451  1980-01-01 00:00   Gerber_TopLayer.GTL
     3403  1980-01-01 00:00   Gerber_TopSolderMaskLayer.GTS
   549358  1980-01-01 00:00   Gerber_TopSilkscreenLayer.GTO
    10226  1980-01-01 00:00   Gerber_TopPasteMaskLayer.GTP
    37386  1980-01-01 00:00   Gerber_BottomLayer.GBL
     2059  1980-01-01 00:00   Gerber_BottomSolderMaskLayer.GBS
    18427  1980-01-01 00:00   Gerber_BottomSilkscreenLayer.GBO
      561  1980-01-01 00:00   Gerber_BoardOutlineLayer.GKO
     1656  1980-01-01 00:00   Drill_PTH_Through.DRL
      636  1980-01-01 00:00   Drill_PTH_Through_Via.DRL
---------                     -------
   635163                     10 files```

So in summary:
- Has anything been lost by deleting the listed files and by (slightly) shortening the existing remaining files
- Why wasn't the image added?
- Could the problem be that EasyEDA uses a different Gerber file format and/or layer naming/id than KiKaid?
- What do I need to do to make this work for EasyEDA Gerber files in general and my case in specific?

Note the warnings I got were as follows:
Gerbalyze Template:
```INFO:root:Layer top copper
INFO:root:    Corresponding overlay layer not found. Skipping.
INFO:root:Layer top mask
INFO:root:    Corresponding overlay layer not found. Skipping.
INFO:root:Layer top silk
INFO:root:    Corresponding overlay layer not found. Skipping.
INFO:root:Layer top paste
INFO:root:    Corresponding overlay layer not found. Skipping.
INFO:root:Layer bottom copper
INFO:root:    Corresponding overlay layer not found. Skipping.
INFO:root:Layer bottom mask
INFO:root:    Corresponding overlay layer not found. Skipping.
INFO:root:Layer bottom silk
INFO:root:    Corresponding overlay layer not found. Skipping.
INFO:root:Layer mechanical outline
INFO:root:    Corresponding overlay layer not found. Skipping.
INFO:root:Layer drill plated
INFO:root:    Corresponding overlay layer not found. Skipping.
INFO:root:Layer drill nonplated
INFO:root:    Corresponding overlay layer not found. Skipping.
/gerbolyze/venv/lib/python3.10/site-packages/gerbonara-1.0.2-py3.10.egg/gerbonara/excellon.py:573: SyntaxWarning: /tmp/tmpcar8an7g/input/Drill_PTH_Through.DRL:19 "G90": G90 header statement found after end of header
  warnings.warn(f'{self.filename}:{self.lineno} "{self.line}": {msg}', SyntaxWarning)
/gerbolyze/venv/lib/python3.10/site-packages/gerbonara-1.0.2-py3.10.egg/gerbonara/excellon.py:573: SyntaxWarning: /tmp/tmpcar8an7g/input/Drill_PTH_Through_Via.DRL:15 "G90": G90 header statement found after end of header
  warnings.warn(f'{self.filename}:{self.lineno} "{self.line}": {msg}', SyntaxWarning)
(venv) root@08ef15a8041d:/EasyEDA-Pro/projects# gerbolyze template --bottom Gerber_PCB4_2023-04-18.zip Gerber_PCB4_2023-04-18-template2.svg
/gerbolyze/venv/lib/python3.10/site-packages/gerbonara-1.0.2-py3.10.egg/gerbonara/layers.py:789: UserWarning: Layer "bottom paste" not found. Found layers: top copper, top mask, top silk, top paste, bottom copper, bottom mask, bottom silk, mechanical outline
  warnings.warn(f'Layer "{side} {use}" not found. Found layers: {", ".join(side + " " + use for side, use in self.graphic_layers)}')
/gerbolyze/venv/lib/python3.10/site-packages/gerbonara-1.0.2-py3.10.egg/gerbonara/excellon.py:573: SyntaxWarning: /tmp/tmptpeao1um/input/Drill_PTH_Through.DRL:19 "G90": G90 header statement found after end of header
  warnings.warn(f'{self.filename}:{self.lineno} "{self.line}": {msg}', SyntaxWarning)
/jgerbolyze/venv/lib/python3.10/site-packages/gerbonara-1.0.2-py3.10.egg/gerbonara/excellon.py:573: SyntaxWarning: /tmp/tmptpeao1um/input/Drill_PTH_Through_Via.DRL:15 "G90": G90 header statement found after end of header
  warnings.warn(f'{self.filename}:{self.lineno} "{self.line}": {msg}', SyntaxWarning)```

Gerbalize paste warnings:
```INFO:root:Layer top copper
INFO:root:    Corresponding overlay layer not found. Skipping.
INFO:root:Layer top mask
INFO:root:    Corresponding overlay layer not found. Skipping.
INFO:root:Layer top silk
INFO:root:    Corresponding overlay layer not found. Skipping.
INFO:root:Layer top paste
INFO:root:    Corresponding overlay layer not found. Skipping.
INFO:root:Layer bottom copper
INFO:root:    Corresponding overlay layer not found. Skipping.
INFO:root:Layer bottom mask
INFO:root:    Corresponding overlay layer not found. Skipping.
INFO:root:Layer bottom silk
INFO:root:    Corresponding overlay layer not found. Skipping.
INFO:root:Layer mechanical outline
INFO:root:    Corresponding overlay layer not found. Skipping.
INFO:root:Layer drill plated
INFO:root:    Corresponding overlay layer not found. Skipping.
INFO:root:Layer drill nonplated
INFO:root:    Corresponding overlay layer not found. Skipping.
/gerbolyze/venv/lib/python3.10/site-packages/gerbonara-1.0.2-py3.10.egg/gerbonara/excellon.py:573: SyntaxWarning: /tmp/tmp89rwf6hx/input/Drill_PTH_Through.DRL:19 "G90": G90 header statement found after end of header
  warnings.warn(f'{self.filename}:{self.lineno} "{self.line}": {msg}', SyntaxWarning)
/gerbolyze/venv/lib/python3.10/site-packages/gerbonara-1.0.2-py3.10.egg/gerbonara/excellon.py:573: SyntaxWarning: /tmp/tmp89rwf6hx/input/Drill_PTH_Through_Via.DRL:15 "G90": G90 header statement found after end of header
  warnings.warn(f'{self.filename}:{self.lineno} "{self.line}": {msg}', SyntaxWarning)```

Thanks!

jaseg, 2023-04-19:
> * Some layers in the original Gerber file were deleted (e.g., Gerber_TopAssemblyLayer.GTA, Gerber_DocumentLayer.GDL, Gerber_DrillDrawingLayer.GDD, How-to-order-PCB.txt -- presumably at least the last one is not really relevant)

This is intended behavior. When gerbolyze doesn't recognize a file, it will ignore it and not copy it to the output. This is necessary since some CAD tools export formatting information for Gerber or Excellon files in separate files. If Gerbolyze would ever not recognize such an auxiliary file, and copy it while modifying the Gerber or Excellon file it is referring to, the output would be broken.

My recommendation is to manually copy over these files afterwards. I might add a command line option in the future.

>     * The individual file sizes were all slightly _smaller_ -- hopefully, no essential data was lost

That is slightly surprising, as in my EasyEDA test files gerbolyze's output is larger. Would you mind sharing one of these files before and after so I can have a look?

In general, the output files will not be the same, since Gerbolyze parses and re-writes every layer. Even for layers without changes this is intended behavior since this way we can make sure that the output files' file format settings are all consistent with one another. Otherwise it could happen that you get different precision or even units in gerbolyze'd and copied files.

When serializing, gerbolyze applies "sane defaults" for some settings, normalizing the file format. Using the default settings, the output files should be readable to any board house. I still recommend you closely check the generated files before submitting them to production. I have made a number of boards using Gerbolyze, but there's always a chance of a terrible bug somewhere, and Gerbolyze comes without warranty. Besides your fab's online Gerber viewer, for validation I can recommend gerbv (which gerbolyze itself is tested against). As a "gold standard", you can use ucamco's reference viewer at https://gerber-viewer.ucamco.com/ .

>     * No evidence of an image being added (as all files were smaller than their original counterparts)

This was a bug in the CI pipeline of resvg-wasi. Do  a `pip install -U resvg-wasi`, and you should get the latest version and `gerbolyze paste` should work again.

jaseg, 2023-04-19:
> That would be helpful since presumably some layers like Gerber_TopAssemblyLayer.GTA would be helpful if you want the pcb fab to assemble your board.

Tbh, I might actually add support for assembly layers anyway, since gerbolyze's gerber library supports moving and rotating boards, too, which would require the same operation on the assembly layer.

> Sure. Can you send me a private email as I do not want to post my boards to the world :)

My email is contact@jaseg.de

> Would this then create a problem with unrecognized layers like Gerber_TopAssemblyLayer.GTA

Probably not in this case, since that layer isn't read together with the other layers anyway. Overall easyEDA's files are well-behaved and unlikely to cause issues here, I'm much more worried about tools like Allegro or CR-9000 that can produce very peculiar output.

puterboy, 2023-04-19:
I sent it you the example before/after Gerber files. Thanks

jaseg, 2023-04-19:
Ok, for documentation here's my observations on the files:

* The drill files increased in size by ~10% since gerbolyze exports them with explicit decimal points and more digits.
* The copper layer files increased in size by ~10% since gerbolyze again exports more digits, and uses a slightly more verbose Gerber syntax with interpolation mode commands on their own lines. At the same time, EasyEDA unnecessarily repeats its interpolation mode commands with every statement, which gerbolyze does not.
* The same happens for the solder mask layers, only here the resulting file size is identical.
* On the silkscreen layer there is a notable reduction in size, which I think is caused by gerbolyze removing the repeated interpolation mode commands. In the silkscreen files this has much more of an effect since they contain a very large number of coordinate statements, especially for EasyEDA's rendering of truetype fonts.

In summary, you should of course check the gerbolyze'd files closely before submitting them to your fab, but as far as I can tell, gerbolyze did everything right and the differences are really just down to EasyEDA's handling of the file format.