jaseg/gerbolyze #42: Documentation for svg-flatten CLI incorrect

magborresen, 2023-09-17:
I've just installed the package through pip on OSX Ventura 13.4.1. I was struggling with using the svg-flatten module since the documentation says to call it with 

```console
svg-flatten [options]
```

This, however, results in an error that the module cannot be found. 

Then I looked into the setup.py for the svg-flatten module and found that the script should be called using

```console
wasi-svg-flatten [options]
```

Now it works with no problem.

Should either the documentation or the setup.py file be updated?

magborresen, 2023-09-17:
Never mind, this still does not work. Only **usvg** options are available when calling 

```console
wasi-svg-flatten [options]
```

jaseg, 2023-11-14:
svg-flatten is a native, binary utility. You can call it as `svg-flatten`, but you have to build and install it natively first. There is a webassembly build of svg-flatten wrapped in a python package on PyPI for use with the `gerbolyze` Python package, since that is written in Python and this is the easiest way to distribute it. That build is specifically meant for the backend of `gerbolyze` though, so it doens't have all the options, and being WASM it's also a lot slower. If you want to use svg-flatten standalone, please compile it from source.

closed 2023-11-14 by jaseg