jaseg/gerbolyze #43: svg-flatten stuck on gerbolyze::dehole_polytree()

Altomare, 2023-09-23:
Hi,

When trying to generate gerbers from SVGs for internal PCB layers, I had a case of svg-flatten being stuck forever.
I edited the shape in Inkscape, and I trimmed in order to get the simplest shape that causes the issue.

Looking at the code, it seems gerbolyze::dehole_polytree() is stuck in an infinite loop.
I tried debugging further but I need to familiarize myself more with that code at the moment.

**Command:**
`svg-flatten --format gerber -f flatten_stuck.svg /tmp/test.gbr`

[image]

fstark, 2023-09-23:
The problem seems to be that ``dehole_polytree_worker`` "viable cut" is not viable: at a moment, the first segment of your first hole is pretty much on the border of the outer path, and the ``ctIntersection`` call returns an empty intersection.

At that moment, the algorithm stop making progress.

I don't know how to properly fix that, but checking if the intersection is empty and chosing a different segment seems to work. I am completely unfamiliar with the library, so the following code is very inefficient and computes the ctDifference and the ctIntersection twice.

`diff --git a/svg-flatten/src/svg_geom.cpp b/svg-flatten/src/svg_geom.cpp
index a36c881..c8477fa 100644
--- a/svg-flatten/src/svg_geom.cpp
+++ b/svg-flatten/src/svg_geom.cpp
@@ -112,24 +112,38 @@ static void dehole_polytree_worker(PolyNode &ptree, Paths &out, queue<PolyTree>
             out.push_back(nod->Contour);
 
         } else {
-            /* Do not add children's children, those were handled in the recursive call above */
-            Clipper c;
-            c.AddPath(nod->Contour, ptSubject, /* closed= */ true);
-            for (int k=0; k<nod->ChildCount(); k++) {
-                c.AddPath(nod->Childs[k]->Contour, ptSubject, /* closed= */ true);
-            }
 
-            /* Find a viable cut: Cut from top-left bounding box corner, through two subsequent points on the hole
-             * outline and to top-right bbox corner. */
-            IntRect bbox = c.GetBounds();
-            Path tri = { { bbox.left, bbox.top }, nod->Childs[0]->Contour[0], nod->Childs[0]->Contour[1], { bbox.right, bbox.top } };
-            c.AddPath(tri, ptClip, true);
-
-            c.StrictlySimple(true);
-            /* Execute twice, once for intersection fragment and once for difference fragment. Note that this will yield
-             * at least two, but possibly more polygons. */
-            c.Execute(ctDifference, todo.emplace(), pftNonZero);
-            c.Execute(ctIntersection, todo.emplace(), pftNonZero);
+            for (int ix=0;ix!=nod->Childs[0]->Contour.size();ix++)
+            {
+                /* Do not add children's children, those were handled in the recursive call above */
+                Clipper c;
+                c.AddPath(nod->Contour, ptSubject, /* closed= */ true);
+                for (int k=0; k<nod->ChildCount(); k++) {
+                    c.AddPath(nod->Childs[k]->Contour, ptSubject, /* closed= */ true);
+                }
+
+                /* Find a viable cut: Cut from top-left bounding box corner, through two subsequent points on the hole
+                * outline and to top-right bbox corner. */
+                IntRect bbox = c.GetBounds();
+                Path tri = { { bbox.left, bbox.top }, nod->Childs[0]->Contour[ix], nod->Childs[0]->Contour[ix+1], { bbox.right, bbox.top } };
+                c.AddPath(tri, ptClip, true);
+
+                c.StrictlySimple(true);
+                /* Execute twice, once for intersection fragment and once for difference fragment. Note that this will yield
+                * at least two, but possibly more polygons. */
+                PolyTree p0;
+                PolyTree p1;
+                c.Execute(ctDifference, p0, pftNonZero);// no progress?
+                c.Execute(ctIntersection, p1, pftNonZero);
+
+                if (p1.GetFirst() )
+                {       //  We got an intersection, se we made progress
+                    c.Execute(ctDifference, todo.emplace(), pftNonZero);// no progress?
+                    c.Execute(ctIntersection, todo.emplace(), pftNonZero);
+                    return ;
+                }
+                //  let's try with the next segment of the first hole
+            }
         }
     }
 }
`

fstark, 2023-09-24:
Above patch doesn't fully work because I missed the outer loop (so the return skips part of the contour).

However, by cutting to the center of the holes, we get something that works

```diff
diff --git a/svg-flatten/src/svg_geom.cpp b/svg-flatten/src/svg_geom.cpp
index a36c881..040bbc4 100644
--- a/svg-flatten/src/svg_geom.cpp
+++ b/svg-flatten/src/svg_geom.cpp
@@ -119,10 +119,17 @@ static void dehole_polytree_worker(PolyNode &ptree, Paths &out, queue<PolyTree>
                 c.AddPath(nod->Childs[k]->Contour, ptSubject, /* closed= */ true);
             }
 
-            /* Find a viable cut: Cut from top-left bounding box corner, through two subsequent points on the hole
-             * outline and to top-right bbox corner. */
+            /* Find a viable cut: Cut from top-left bounding box corner to the horizontal middle of the first hole, down to the bottom and back to the bottom-left */
             IntRect bbox = c.GetBounds();
-            Path tri = { { bbox.left, bbox.top }, nod->Childs[0]->Contour[0], nod->Childs[0]->Contour[1], { bbox.right, bbox.top } };
+
+            cInt sumx = 0;
+            for (auto &v:nod->Childs[0]->Contour)
+            {
+                sumx += v.X;
+            }
+            sumx /= nod->Childs[0]->Contour.size();
+
+            Path tri = { { bbox.left, bbox.top }, { sumx, bbox.top }, { sumx,bbox.bottom }, {bbox.left, bbox.bottom} };
             c.AddPath(tri, ptClip, true);
 
             c.StrictlySimple(true);
@@ -130,6 +137,9 @@ static void dehole_polytree_worker(PolyNode &ptree, Paths &out, queue<PolyTree>
              * at least two, but possibly more polygons. */
             c.Execute(ctDifference, todo.emplace(), pftNonZero);
             c.Execute(ctIntersection, todo.emplace(), pftNonZero);
+
+            if (!todo.back().GetFirst())
+                printf( "***** ERROR INTERSECTION EMPTY *****\n" );
         }
     }
 }
```

I also added a message in case of empty intersection -- which was the original infinite loop issue.

closed 2023-11-14 by jaseg

jaseg, 2023-11-14:
Thank y'all for the bug report and the suggested fix! I have pushed a fix to main, which will land in a release in the coming weeks.

In the end, the fix was a bit more complicated than suggested by @fstark. The infinite loop did not come from empty polytrees being pushed to the stack, as those would just be discarded by the next worker call. Instead, the infinite loop arose from the worker ceasing to make progress under particular circumstances: It would get a polytree with one hole in, and push a new polytree with the same hole back on the stack.

The issue ended up being the construction of the "tool" polygon. This was supposed to be a convex polygon, with two points at the top left and right corners of the bounding box, and two corners at the ends of an arbitrary segment of the hole's outline polygon. The issue with this was that during construction, the orientation of that hole outline segment was never checked, and so it could happen that sometimes, it would create a hourglass-shaped tool polygon with a big, fat self-intersection close to the hole. If at the same time we were unlucky enough for clipper to return the hole polygon with its first outline point in the same spot as during our last call, that would lead to this infinite loop.

The fix ended up being pretty simple: We just look through the hole polygon's outline segments until we find one that is not exactly vertical, and orient it to make sure we don't get a self-intersection.

The remaining TODO is to add a regression test for this.

commit 7a80ac76 (2023-11-14): Fix infinite loop bug in dehole_polytree

Closes #43. Thanks to github users @Altomare for reporting this one, and
@fstark for suggesting a fix.
commit 92277733 (2023-11-14): Fix infinite loop bug in dehole_polytree

Closes #43. Thanks to github users @Altomare for reporting this one, and
@fstark for suggesting a fix.
commit 8e8bcee2 (2023-11-14): Fix infinite loop bug in dehole_polytree

Closes #43. Thanks to github users @Altomare for reporting this one, and
@fstark for suggesting a fix.