jaseg/gerbolyze #47: Tests fail with resvg 0.42.0

Wulfsta, 2024-06-22:
Resulting error from a build attempt:
```
DEBUG:root:svg-flatten args: --format gerber-outline --precision 6 --no-usvg --trace-space 0.1 --only-groups g-drill-nonplated /build/tmp2r283dcf.svg /build/tmpau6wk_fg.gbr
--- Logging error ---
Traceback (most recent call last):
  File "/nix/store/6b1fqdwb3g56j5pazv8zkx9qd0mv3wiz-python3-3.11.9/lib/python3.11/logging/__init__.py", line 1110, in emit
    msg = self.format(record)
          ^^^^^^^^^^^^^^^^^^^
  File "/nix/store/6b1fqdwb3g56j5pazv8zkx9qd0mv3wiz-python3-3.11.9/lib/python3.11/logging/__init__.py", line 953, in format
    return fmt.format(record)
           ^^^^^^^^^^^^^^^^^^
  File "/nix/store/6b1fqdwb3g56j5pazv8zkx9qd0mv3wiz-python3-3.11.9/lib/python3.11/logging/__init__.py", line 687, in format
    record.message = record.getMessage()
                     ^^^^^^^^^^^^^^^^^^^
  File "/nix/store/6b1fqdwb3g56j5pazv8zkx9qd0mv3wiz-python3-3.11.9/lib/python3.11/logging/__init__.py", line 377, in getMessage
    msg = msg % self.args
          ~~~~^~~~~~~~~~~
TypeError: not all arguments converted during string formatting
Call stack:
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/build/source/gerbolyze/__main__.py", line 2, in <module>
    gerbolyze.cli()
  File "/nix/store/x945cn634yr39bbjz5rv7my5yj0gx6ry-python3.11-click-8.1.7/lib/python3.11/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/nix/store/x945cn634yr39bbjz5rv7my5yj0gx6ry-python3.11-click-8.1.7/lib/python3.11/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/nix/store/x945cn634yr39bbjz5rv7my5yj0gx6ry-python3.11-click-8.1.7/lib/python3.11/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/nix/store/x945cn634yr39bbjz5rv7my5yj0gx6ry-python3.11-click-8.1.7/lib/python3.11/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/nix/store/x945cn634yr39bbjz5rv7my5yj0gx6ry-python3.11-click-8.1.7/lib/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/build/source/gerbolyze/__init__.py", line 97, in paste
    overlay_grb = svg_to_gerber(processed_svg.name, no_usvg=True,
  File "/build/source/gerbolyze/__init__.py", line 584, in svg_to_gerber
    logging.debug('using svg-flatten at', candidate)
Message: 'using svg-flatten at'
Arguments: ('svg-flatten',)
```

jaseg, 2024-06-22:
I'm able to reproduce this. I'll have a look.

closed 2024-06-26 by jaseg

Wulfsta, 2024-07-09:
@jaseg I think this fix might have introduced an issue where `empty-template` does not fill in the `viewBox` fields and now generates a lot of spurious noise. It all works as expected, just introduced warnings on Gerbolyze generated templates.

jaseg, 2024-07-10:
@Wulfsta Can you post what `empty-template` gives you? The `viewBox` attr is hardcoded [here](https://github.com/jaseg/gerbolyze/blob/main/gerbolyze/__init__.py#L506) and it would be weird if it was missing.

v3.1.8 that fixed this issue included some extra checks on `viewBox` that were a bit broken and produced harmless but noisy command-line output. v3.1.9 fixes these checks and should be less noisy.

Wulfsta, 2024-07-10:
Oh, that's odd... the generated svg looks fine:
```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
        <svg version="1.1"
           xmlns="http://www.w3.org/2000/svg"
           xmlns:xlink="http://www.w3.org/1999/xlink"
           xmlns:inkscape="http://www.inkscape.org/namespaces/inkscape"
           xmlns:sodipodi="http://sodipodi.sourceforge.net/DTD/sodipodi-0.dtd"
           width="100.0mm" height="100.0mm" viewBox="0 0 100.0 100.0" >
          <defs/>
          <sodipodi:namedview inkscape:current-layer="g-top-copper" />
          <g id="g-top-paste" inkscape:label="top paste" inkscape:groupmode="layer"></g>
  <g id="g-top-silk" inkscape:label="top silk" inkscape:groupmode="layer"></g>
  <g id="g-top-mask" inkscape:label="top mask" inkscape:groupmode="layer"></g>
  <g id="g-top-copper" inkscape:label="top copper" inkscape:groupmode="layer"></g>
  <g id="g-bottom-copper" inkscape:label="bottom copper" inkscape:groupmode="layer"></g>
  <g id="g-bottom-mask" inkscape:label="bottom mask" inkscape:groupmode="layer"></g>
  <g id="g-bottom-silk" inkscape:label="bottom silk" inkscape:groupmode="layer"></g>
  <g id="g-bottom-paste" inkscape:label="bottom paste" inkscape:groupmode="layer"></g>
  <g id="g-mechanical-outline" inkscape:label="mechanical outline" inkscape:groupmode="layer"></g>
  <g id="g-drill-plated" inkscape:label="drill plated" inkscape:groupmode="layer"></g>
  <g id="g-drill-nonplated" inkscape:label="drill nonplated" inkscape:groupmode="layer"></g>
  <g id="g-other-comments" inkscape:label="other comments" inkscape:groupmode="layer"></g>
        </svg>
```
But running `gerbolyze convert test.svg test` produces:
```
No viewBox given on <svg> root, using width/height attributes.
Resulting page width 100 mm x 100 mm
Resulting document scale 1 x 1
No viewBox given on <svg> root, using width/height attributes.
Resulting page width 100 mm x 100 mm
Resulting document scale 1 x 1
No viewBox given on <svg> root, using width/height attributes.
Resulting page width 100 mm x 100 mm
Resulting document scale 1 x 1
No viewBox given on <svg> root, using width/height attributes.
Resulting page width 100 mm x 100 mm
Resulting document scale 1 x 1
No viewBox given on <svg> root, using width/height attributes.
Resulting page width 100 mm x 100 mm
Resulting document scale 1 x 1
No viewBox given on <svg> root, using width/height attributes.
Resulting page width 100 mm x 100 mm
Resulting document scale 1 x 1
No viewBox given on <svg> root, using width/height attributes.
Resulting page width 100 mm x 100 mm
Resulting document scale 1 x 1
No viewBox given on <svg> root, using width/height attributes.
Resulting page width 100 mm x 100 mm
Resulting document scale 1 x 1
No viewBox given on <svg> root, using width/height attributes.
Resulting page width 100 mm x 100 mm
Resulting document scale 1 x 1
No viewBox given on <svg> root, using width/height attributes.
Resulting page width 100 mm x 100 mm
Resulting document scale 1 x 1
No viewBox given on <svg> root, using width/height attributes.
Resulting page width 100 mm x 100 mm
Resulting document scale 1 x 1
No viewBox given on <svg> root, using width/height attributes.
Resulting page width 100 mm x 100 mm
Resulting document scale 1 x 1
```

jaseg, 2024-07-11:
Oh, meh. I think usvg changed that recently. I'll figure out a workaround at some point, for now please feel free to just ignore the warnings.

commit 6f075fff (2024-06-26): svg-flatten: Fix handling of missing viewBox

This makes svg-flatten compatible with usvg v0.42.0, which changed its
viewbox/width/height output.

Closes #47