jaseg/gerbolyze #50: wasi-svg-flatten subprocess error

Mr-V1ct0r, 2024-12-14:
I've been racking my brain for the last day trying to find ways to convert my adobe illustrator eurorack panel design to KiCad format.  After scouring the net and trying a few diff options, I found Gerbolyze through alorman's github page.  I followed his directions and the directions here but I'm out of my league with this python, github, rust, visual studio, cargo etc etc.  I've been working on this for 4 hours solid jumping through hoops to get to the point I'm at, and I feel like I'm so damn close to getting to where I need to be with this.  I'm finally able to actually run the script and I'm getting this error.  Please help me resolve this issue.  Gerbolyze is my last hope!!!  

Thanks,

-Victor

```
python C:\Users\v1cto\AppData\Roaming\Python\Python313\Scripts\wasi-svg-flatten.exe --format kicad --sexp-mod-name Amoeba_Panel_4-KiCad C:\Users\v1cto\Documents\KiCad\Amoeba\Amoeba Panel\Amoeba Panel v1-FINAL-FLAT-4-KiCad-01.svg C:\Users\v1cto\Documents\KiCad\Amoeba\Amoeba Panel\Amoeba Panel v1-FINAL-FLAT-4-KiCad.kicad_mod
['C:\\Users\\v1cto\\Panel', 'C:\\Users\\v1cto\\AppData\\Local\\Temp\\tmpjxktpips']
Error: Operation not permitted (os error 63).
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\v1cto\AppData\Roaming\Python\Python313\Scripts\wasi-svg-flatten.exe\__main__.py", line 7, in <module>
    sys.exit(run_svg_flatten())
             ~~~~~~~~~~~~~~~^^
  File "C:\Users\v1cto\AppData\Roaming\Python\Python313\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "C:\Users\v1cto\AppData\Roaming\Python\Python313\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "C:\Users\v1cto\AppData\Roaming\Python\Python313\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\v1cto\AppData\Roaming\Python\Python313\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\v1cto\AppData\Roaming\Python\Python313\site-packages\svg_flatten_wasi\__init__.py", line 143, in run_svg_flatten
    run_usvg(input_file, f.name, **usvg_args)
    ~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\v1cto\AppData\Roaming\Python\Python313\site-packages\svg_flatten_wasi\__init__.py", line 111, in run_usvg
    res = subprocess.run([candidate, *args], check=True)
  File "C:\Users\v1cto\AppData\Local\Programs\Python\Python313\Lib\subprocess.py", line 577, in run
    raise CalledProcessError(retcode, process.args,
                             output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['wasi-usvg', 'C:\\Users\\v1cto\\Panel', 'C:\\Users\\v1cto\\AppData\\Local\\Temp\\tmpjxktpips']' returned non-zero exit status 1.
```

Wulfsta, 2024-12-18:
`Error: Operation not permitted (os error 63).` usually indicates a permissions issue - make sure the user executing this can read the files, and write to the directories.

Techmokid, 2025-01-05:
Actually I'm having the exact same fault. Tried on multiple machines, all failing. It seems to directly be the cause of "wasi-usvg" itself, as running "wasi-usvg in.svg out.svg" is throwing the same fault. Unsure if wasi-usvg itself is corrupted or something, but it's installed at "C:\Users\aj200\AppData\Local\Programs\Python\Python3xx\Scripts" when installing the dependancy "resvg-wasi".

I just cannot get this to play nicely nomatter what I do. Even running direct from administrator commandline, it still doesn't like it. This is directly a wasi-usvg error, not a gerbolyze error. I've tried using administrator commandline, reinstalling, and everything