jaseg/gerbolyze #51: Version bump

Wulfsta, 2024-12-18:
@jaseg could you please bump the version? I see there were some commits, and assume there should be a new tag. I [rely on tags](https://github.com/NixOS/nixpkgs/blob/6083c9e1ff258d1ccd6c7f96a4ab16f0ac8dcbeb/pkgs/by-name/ge/gerbolyze/package.nix#L92) here and in Gerbonara to autoupdate these packages downstream in NixOS.

If you are planning to update without release tags, I can also switch to tracking this in an unstable fashion, and just use the latest commit.

jaseg, 2024-12-21:
Hi there,

I'll put out a new release soon-ish. I just got stuck mid-work because of some IRL distractions, and the new changes aren't tested properly yet :)

closed 2024-12-21 by jaseg

Wulfsta, 2024-12-21:
My apologies for the incorrect assumption then!

jaseg, 2024-12-21:
no problem :)